ihrapsa/KlipperWrt #1: OpenWrt Firmware sysupgrade not factory

morganfw, 2021-03-16:
Hello,
the firmware contained in https://github.com/ihrapsa/KlipperWrt/tree/main/Firmware/OpenWrt_snapshot is a *sysupgrade.bin not a *factory.bin, so I'm not able to flash on original Creality WB-01 through TF card.

Also could you kindly publish also the .config file you used to compile OpenWrt?

Thanks for the great work you did.

ihrapsa, 2021-03-16:
The box has a U-Boot loader with a recovery procedure that let's you flash any bin: stock, factory or sysupgrade. The steps are written at the end of the guide as well but I'll present them here as well: To flash the sysupgrade file rename it to `root_uImage` put it on a USB flash -> plug it in the box powered off ->press and hold reset button -> power on the box while holding the button -> keep pressing ~10 seconds -> release and wait a couple of minutes -> plug an ethernet cable and ssh into it at 192.168.1.1

ihrapsa, 2021-03-16:
What printer do you have? (If you flash my image it  might not detect the printer if your mainboard's serial to USB convertor is not ch341 - it's a kmod and until the box gets fully supported you'll have to build any kmods you need ). You can find the list with all the kmods you need at step 1. I can generate you a .config file if you really need but I build my images by selecting only those kmods in `make menuconfig` and it might be faster to try it yourself. :blush:

ihrapsa, 2021-03-16:
Here it is anyways :grin: [config.zip](https://github.com/ihrapsa/KlipperWrt/files/6152592/config.zip)

morganfw, 2021-03-17:
Hi and thank you for answer.
Installed your firmware through USB flash, and I also tried to compile my own with LuCI interface and other little modifications, so, thank you for the config file.

I own an SKR v1.3 and SKR v1.4 Turbo boards, and a Tronxy X5SA Pro Printer. Searching the Internet doesn't help me to find if they are CH341 or FTDI boards.
You know what kind of USB to serial are present on that boards? I can try to compile with both kmods, or only one allowed?

Thanks

ihrapsa, 2021-03-17:
Easiest way to find out is to connect each board to a Linux PC(even a rpi should work) and run `sudo dmesg`. The log should tell you what driver it uses. You can build the image with all the USB to serial drivers kmods. Just make sure to stay under 16mb with the image.

morganfw, 2021-03-17:
Thank you for all answers, I will try and test.
I close the "issue".

Regards

closed 2021-03-17 by morganfw