ihrapsa/KlipperWrt #3: Can’t open fluidd in safari

sarcela, 2021-04-27:
Hi, first of all I want to say thank you, because of this I can finally use my useless box, but i have two issues, the first one is that I can’t open fluidd in my iPad, I was reading a lot but with any answer, but i can open the duet-web-control, so I think that is a fluidd problem with the msie6 browser or something like that, and the other issue is that I can’t run the autobed leveling, do you know what I am doing wrong? Thank you very much another time, this is awesome!

ihrapsa, 2021-04-27:
Hi, yeah fluidd is browser dependent. I don't have an iPad but I've heard people having trouble with older iOS. I had similar issues on Android 4.1.1 on chrome but switched to a different browser that still received updates and it worked. Also are you sure using `http://` and not `https://`?

I see you also set up duet-web-control, although it shouldn't be a problem I wouldn't recommend having both Moonraker (with fluidd) and dwc running on the box at the same time. It would eat up a few cpu percents for free. so better stick with one or the other. you can always ` service dwc disable` and it won't automatically boot. You can start it or enable it anytime you want.

Regarding auto bed levelling I just finished setting up mine the other day so I don't think it's an issue related to the box. what exactly is the issue? What probe are you using?

sarcela, 2021-04-27:
Hi, thank you for your answer, I m sorry for the time, yes, I tried everything, http, https, hostname and ip and no, all I get is forbidden by nginx,, I try to put in some files “keepalive_disable none;” but fluidd stop working.

And for the other thing, how do you get bed leveling work? I get unknown command or something like that in the console

ihrapsa, 2021-04-27:
If you're getting forbidden by Nginx it's probably because safari automatically switches to `https` even though you typed `http`. Are you able to get into fluidd on desktop?

Regarding bed mesh, did you follow this [guide](https://github.com/KevinOConnor/klipper/blob/master/docs/Bed_Mesh.md)? What command did you type in console? Aren't you using the `Tune` section in fluidd to calibrate bed mesh? Are you using blotch?

sarcela, 2021-04-27:
Yes, i am able to open fluidd In chrome, on my desktop, but I want to open it in my iPad, even in my cellphone, but i can’t, I am going to look if I could get the http and not https, thank you very much.
Let me see that guide, but yes, I was trying to do de bed mesh with the tune section in the printer.
I will try that and I will tell you what is going on.
Thank you.

ihrapsa, 2021-04-27:
Np. Just ping me on discord if you have other problems. :)

closed 2021-04-28 by ihrapsa

sarcela, 2022-04-23:
i find the solution of this by modifing the nginx file

/etc/config/nginx

ijust changed 80 to 82

config server '_redirect2ssl'
	list listen '82'
	list listen '[::]:82'
	option server_name '_redirect2ssl'
	option return '302 https://$host$request_uri'