ihrapsa/KlipperWrt #4: How to cross-compile a c program for creality box?

Rbn3D, 2021-04-28:
Hi, I'm trying to install octoprint on my box by following your guide and this one:

https://community.octoprint.org/t/assorted-guides-as-formerly-maintained-on-the-github-wiki/21146/8

The problem is that some octoprint dependencies, like "regex" (it's a pip package) requires a newer version of ggc in order to be built and installed.

My box has gcc 7.4 (from opkg repo) and this requires, at least, 8.4.

Not sure if it's better to cross-compile gcc itself, or just the dependencies that have no suitable wheel and cannot be compiled under gcc 7.4.

Since creality box version of OpenWRT is not official (yet), not sure if cross-compile it's even suitable there... I would greatly appreciate any hints on this...

Thanks in advance 👍

ihrapsa, 2021-04-28:
Hi, yeah, regex is a `Cython` package that needs to be cross compiled. I followed that guide as well when I tried octoprint on this box. I do not recommend it though since it's very power intensive and printjobs ended up with jerks and blobs... check the test print in the [picture ](https://github.com/ihrapsa/KlipperWrt/blob/main/screenshots/test_print.jpg )of my repo. If you want to experiment yourself, openwrt image builder makes that easy for you: Check the last line (38) in the [requirements.txt](https://github.com/ihrapsa/KlipperWrt/blob/main/requirements.txt) file. I explained there the steps to cross compile python packages that can't be built on the box. But again, if you manage to get octoprint running it won't be decently usable

Rbn3D, 2021-04-28:
Yes, I know that printing from octoprint wouldn't be usable, but by using octoprint I can monitor SD prints and send gcode easily, without having to flash klipper...

Ok I will try to follow these steps, thanks!

ihrapsa, 2021-04-28:
Yeah, that's a good use. Note that sending gcode to printer sd card through octoprint takes ages. It's using serial and not a dedicated file transfer protocol. Also most of octoprint functionality is not available with SD prints. Good luck! :)

ihrapsa, 2021-04-28:
If you're considering it, I really reccomend changing to klipper with fluidd or mainsail. It's faster, easier to tweak, light on CPU, slicer upload, thumbnails, you can also plug a usb webcam and stream with no issues on this box. :)

closed 2021-04-28 by ihrapsa