ihrapsa/KlipperWrt #5: Can't connect using to box using Putty over wifi

Kruze17, 2021-05-28:
[image]

Hello.
I have successfully followed and installed openwrt on my box and configured all /etc/config/xxxxx folders using rj-45 cable over ethernet. It connects to my router also. But i am not able to connect it over wifi using putty. It shows "Network error: Connection timed out".

Any heads up on what i am doing or going wrong?
Thanks.

ihrapsa, 2021-05-28:
are you sure the box gets an ip from your router DHCP?

Kruze17, 2021-05-29:
[image]

Yes i am sure.
For better confirmation i downloaded Nirsoft's wireless network watcher, which provides better information than my router's login page.

IP Address	Device Name	MAC Address	Network Adapter Company	Device Information	User Text	First Detected On	
192.168.1.5		FC-EE-E6-00-82-98	FORMIKE ELECTRONIC CO., LTD			29-05-2021 12:27:41	29-05-2021 

Please check the picture also.

ihrapsa, 2021-05-29:
Your router DHCP wifi list should be more than enough. There should be an OpenWrt entry and it's assigned ip address. Is the above screenshot with the Ethernet disconnected?

Kruze17, 2021-05-29:
> Your router DHCP wifi list should be more than enough. There should be an OpenWrt entry and it's assigned ip address. Is the above screenshot with the Ethernet disconnected? "Formike Electronics" seems to be a lcd manufacturing company. That's odd

YES! Ethernet is disconnected from creality wifi box. Am i supposed to keep it connected?
I am following this guide: https://letsprint3d.net/install-klipper-on-creality-wifi-box/
It tells to unplug the ethernet cable after rebooting.
[image]

Is this guide wrong?

ihrapsa, 2021-05-29:
that guide is ok for that section though I highly recommend following the github guide in parallel as well since it's updated more frequently. we might move the conversation to discord as well.

Kruze17, 2021-05-29:
ok sure.

closed 2021-05-29 by Kruze17