ihrapsa/KlipperWrt #6: Cannot install nginx ssl with openwrt snapshot image for creality box

Aimen06, 2021-05-29:
Hi,
i have tried to install package nginx-ssl with no luck.
I  have an error :
`Unknown package 'nginx-ssl'.
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for nginx-ssl found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package nginx-ssl.`

Here some info.
`uname -a
Linux OpenWrt 5.4.105 #0 Fri Mar 19 23:37:51 2021 mips GNU/Linux`

`cat /proc/cpuinfo
system type		: MediaTek MT7688 ver:1 eco:2
machine			: Creality WB-01
processor		: 0
cpu model		: MIPS 24KEc V5.5
BogoMIPS		: 385.02
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 32
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16 dsp
Options implemented	: tlb 4kex 4k_cache prefetch mcheck ejtag llsc pindexed_dcache userlocal vint perf_cntr_intr_bit perf
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 0
VCED exceptions		: not available
VCEI exceptions		: not available`

ihrapsa, 2021-05-29:
The firmware got updated to a new kernel version. Sysupgrade to it. ;)

Aimen06, 2021-05-29:
Ok thanks i am do it 👍

closed 2021-05-29 by ihrapsa