ihrapsa/KlipperWrt #12: [dev] consider zram / zswap

ihrapsa, 2021-06-24:
This is a kind of swap that resides on RAM in a compressed form. It will make swapping and ram usage more efficient and it will increase the lifespan of the sd card as well.

Resources:
https://openwrt.org/packages/pkgdata/zram-swap
https://openwrt.org/packages/pkgdata/kmod-zram
https://forum.openwrt.org/t/retracted-compressed-memory-this-project-is-officially-retracted-take-note-about-its-future/49076

thomsonOZ, 2021-09-06:
Updated to my KlipperWRT, will review the performance in next few days.

ihrapsa, 2021-09-06:
> Updated to my KlipperWRT, will review the performance in next few days.

hmm... Thanks, but actually I configured KlipperWrt as default with  `zram-swap`. I tried experimenting with different compression algorithms and ram sizes as well as zswap (which seems to be a more efficient way of using ram) but that `zram-swap` configures everything depending on the amount of ram you have and stuck with that. If you manage to get better performance with a different configuration please share 🙂