ihrapsa/KlipperWrt #18: [guide] DSLR snapshots - gphoto2 triggered

ihrapsa, 2021-07-20:
Use `gphoto2` to trigger a **DSLR** shutter. With this software you can also transfer pictures to the Linux controlling device as well. Luckily this package is available inside OpenWrt repo:
`opkg update && opkg install gphoto2`

Connection:

         📷 <----- USB cable -----> 📥
    (DSLR camera)             (OpenWrt box)

**video tutorial:**
https://www.youtube.com/watch?v=1eAYxnSU2aw

**Maybe useful for my D3300:**
https://github.com/gphoto/libgphoto2/blob/master/camlibs/ptp2/cameras/nikon-d3300.txt

wbatty, 2021-09-03:
Stumbled across this while doing a google search for how to get DSLRs working with klipper, I had totally forgotten about the shell command extension.. Anyway I made a macro that implements parking the toolhead, maybe someone will find it useful:

```
[gcode_shell_command gphoto2_trigger_snapshot]
command: gphoto2 --auto-detect --trigger-capture
timeout: 10. ; make sure this is long enough to account for focus time
verbose: False

[gcode_macro TAKE_SNAPSHOT]
description: Saves current location, moves toolhead to rear of bed, and takes a snapshot before returning.
gcode:
  {% set delayMS = 2000 %} ; Delay after moving and before taking the snapshot, deals with buffering so we don't get the print head in motion
  {% set snapshotPositionX = printer.toolhead.axis_maximum.x / 2 %} ; Middle of bed
  {% set snapshotPositionY = printer.toolhead.axis_maximum.y - 5 %} ; rear of bed w/ some buffer
  {% set currentX = printer.gcode_move.gcode_position.x %}
  {% set currentY = printer.gcode_move.gcode_position.y %}
  {% set currentZ = printer.gcode_move.gcode_position.z %} ; could be used for z-hop which we are not doing in this macro, so it's best to run this in AFTER_LAYER_CHANGE
  SAVE_GCODE_STATE NAME=take_snapshot_state

  {% if printer.extruder.can_extrude|lower == 'true' %}
    G10 ; retract
  {% endif %}

  G90

  G0 X{snapshotPositionX} Y{snapshotPositionY} F5000.0 ;Move to snapshot position

  G4 P{delayMS} ; Dwell for delayMS seconds

  RUN_SHELL_COMMAND CMD=gphoto2_trigger_snapshot

  G0 X{currentX} Y{currentY} F5000.0 ;Return to original position

  {% if printer.extruder.can_extrude|lower == 'true' %}
    G11 ; unretract
  {% endif %}

  RESTORE_GCODE_STATE NAME=take_snapshot_state

````

ihrapsa, 2021-09-03:
Woa this is awesome, thanks for sharing! ☺

Twentse3d, 2022-11-16:
For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt as follows:

Please make sure there is sufficient quoting around the arguments.

pi@speeder-pad:~$
pi@speeder-pad:~$     env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --get-config capturetarget

*** Error ***
An error occurred in the io-library ('I/O problem'): No error description available

*** Error ***
An error occurred in the io-library ('I/O problem'): No error description available
*** Error (-7: 'I/O problem') ***

?????? whats problem ?

ihrapsa, 2022-11-16:
Please explain your issue properly first.

Twentse3d, 2022-11-16:
whats problem ???

pi@speeder-pad:~$ gphoto2 --auto-detect
Model                          Port
----------------------------------------------------------
Canon EOS M50                  usb:001,006
pi@speeder-pad:~$ gphoto2 --get-config capturetarget

*** Error ***
An error occurred in the io-library ('I/O problem'): No error description available

*** Error ***
An error occurred in the io-library ('I/O problem'): No error description available
*** Error (-7: 'I/O problem') ***

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --get-config capturetarget

Please make sure there is sufficient quoting around the arguments.

pi@speeder-pad:~$ gphoto2 --set-config capturetarget=1

*** Error ***
An error occurred in the io-library ('I/O problem'): No error description available

*** Error ***
An error occurred in the io-library ('I/O problem'): No error description available
*** Error (-7: 'I/O problem') ***

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --set-config capturetarget=1

Please make sure there is sufficient quoting around the arguments.

pi@speeder-pad:~$

ihrapsa, 2022-11-16:
> whats problem ???

Please state your issue properly. I don't know anything about your setup, what you did to try to fix it, when those errors happen and so on. And please format any logs you have properly between "```" marks or click on the `Add code` button.

Xapu1337, 2023-09-24:
> whats problem ???
> 
> ## pi@speeder-pad:~$ gphoto2 --auto-detect
> Model Port
> Canon EOS M50 usb:001,006 pi@speeder-pad:~$ gphoto2 --get-config capturetarget
> 
> *** Error *** An error occurred in the io-library ('I/O problem'): No error description available
> 
> *** Error *** An error occurred in the io-library ('I/O problem'): No error description available *** Error (-7: 'I/O problem') ***
> 
> For debugging messages, please use the --debug option. Debugging messages may help finding a solution to your problem. If you intend to send any error or debug messages to the gphoto developer mailing list [gphoto-devel@lists.sourceforge.net](mailto:gphoto-devel@lists.sourceforge.net), please run gphoto2 as follows:
> 
> ```
> env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --get-config capturetarget
> ```
> 
> Please make sure there is sufficient quoting around the arguments.
> 
> pi@speeder-pad:~$ gphoto2 --set-config capturetarget=1
> 
> *** Error *** An error occurred in the io-library ('I/O problem'): No error description available
> 
> *** Error *** An error occurred in the io-library ('I/O problem'): No error description available *** Error (-7: 'I/O problem') ***
> 
> For debugging messages, please use the --debug option. Debugging messages may help finding a solution to your problem. If you intend to send any error or debug messages to the gphoto developer mailing list [gphoto-devel@lists.sourceforge.net](mailto:gphoto-devel@lists.sourceforge.net), please run gphoto2 as follows:
> 
> ```
> env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --set-config capturetarget=1
> ```
> 
> Please make sure there is sufficient quoting around the arguments.
> 
> pi@speeder-pad:~$

maybe late but it is the USB port you plug it in!
if its a dock or the pi just can't provide enough power to the USB ports for it to work.