ihrapsa/KlipperWrt #22: klipperwrt.local not known

fenyvesi, 2021-08-13:
Nice project, thanks.
After setting up the WIFI Client by the instructions of V2.1 Release Notes I can connect with the IP given by DHCP (192.168.0.211).

The http://klipperwrt.local:81 seems not to work, klipperwrt.local is not known by the network.
How to set it up ?

Thanks.
George

ihrapsa, 2021-08-13:
Hey, 
1. Have you tried a different browser? 
2. What OS are you on? 
3. What method did you use to setup KlipperWrt?

ihrapsa, 2021-08-13:
Wait, dumb me. I forgot to edit the Release notes instructions. The ones from README were updated. You will be able to access KlipperWrt.local only after you install the second script. That's because there's no `avahi daemon` preinstalled. The second script will install it for you

fenyvesi, 2021-08-13:
You were extra fast.

The answers:
    Have you tried a different browser? 
              I tried ping, not only browser.
    What OS are you on?
         WIN10
    What method did you use to setup KlipperWrt?
              Nothing outside your description in v2.1

New development is that after rebooting the box I can't connect the luci interface with the DHCP IP address that has worked the 1st time.

ihrapsa, 2021-08-13:
> You were extra fast.
>
> The answers:
> Have you tried a different browser?
> I tried ping, not only browser.
> What OS are you on?
> WIN10
> What method did you use to setup KlipperWrt?
> Nothing outside your [...]

I would recommend following the `Installing Script method` under `Automatic steps`. To start fresh, just hold the reset button for 6-10 sec and OpenWrt will reset to the moment you flashed it.

fenyvesi, 2021-08-13:
The answers crossed each other.
I try by your advices and going back.

**I would recommend following the Installing Script method under Automatic steps. To start fresh, just hold the reset button for 6-10 sec and OpenWrt will reset to the moment you flashed it. Then follow the steps to setup wifi and execute scripts no. 1 and then 2**

ihrapsa, 2021-08-13:
Sure. 

I think the hostname.local issue has been resolved so I can close this ticket. 

If you need extra guidance you could also join the discord server :)

closed 2021-08-13 by ihrapsa

fenyvesi, 2021-08-13:
Unfortunately the problem does exist yet.

The steps I made in installation. I used eth0 directly connected to a notebook with static IP..

1. The V2.1 FluidWRT-Factory..... file copied to SD card, restart.
2. sysupgrade with v2.1 FluidWrt_SYSUPGRADE.... file. reboot.
3. Setting wireless.
4. Trying with the DHCP given IP through WIFI. It works only once, klipperwrt.local doesn't exist. After rebooting the DHCP IP (192.168.0.211:81) flashes the screen with text "LUA-Lua Configuration Interface", the adress changes to .../cgi-bin/luci and the connection fails.
5. Downloading and running install.sh. It works. Reboot.
6. The luci interface doesn't work through WIFI connection (see above), no klipperwrt.local.
7. It works through eth0 with 191.168.1.1 only.

I attach the syslog file.

[syslog.txt](https://github.com/ihrapsa/KlipperWrt/files/6983493/syslog.txt)

Thanks.

ihrapsa, 2021-08-13:
4. klipperwrt.local should only work after you execute the scripts (the scripts will install the avahi-daemon)

Looking at your syslog it appears you are on the same subnet as the box (192.168.1.100)which might be conflicting with the box. So I would recommend changing the box's ip to 192.168.2.1 or 192.168.3.1 (That would be under Network -> Interfaces -> change the static ip)

ALSO, Try using the `Installing script method`  with the two scripts instead. The Extroot method seems to be a bit buggy on my tests.

fenyvesi, 2021-08-14:
I installed by the 2-scripts method. The result is the same. 
The root of the problem is not the avahi but the not-working wifi client and the result is the missiing klipperwrt.local.

After starting the box I can ping the wlan DHCP given IP: 192.168.0.211.
If I try to connect it (192.168.0.211:81) it connects, flashes the screen with text "LUA-Lua Configuration Interface", the address changes to .../cgi-bin/luci and the connection fails.

I tried with 2 different APs, the result is the same.

In the meantime the syslog has some strange messages:
     ... daemon.notice wpa_supplicant[1420]: wlan0: Unknown event 37
    .... daemon.info dnsmasq[1259]: exiting on receipt of SIGTERM

    ... daemon.notice wpa_supplicant[1420]: wlan0: Unknown event 37
    ... daemon.notice wpa_supplicant[1420]: wlan0: CTRL-EVENT-BEACON-LOSS

[syslog2.txt](https://github.com/ihrapsa/KlipperWrt/files/6986146/syslog2.txt)

ihrapsa, 2021-08-14:
I'm not sure I completely get it. How did you install by the 2 scripts method if you didn't even manage to get wifi client setup? 🤔

Please join the discord server since it's a lot easier for me to help there

fenyvesi, 2021-08-14:
Through Ehernet cable with 192.168.1.1.

ihrapsa, 2021-08-14:
As long as you make the box connect to your wifi network as a client and you make sure its lan static IP subnet is different tham the one of your router, you should be able to connect without issues. I'm not sure what's your exact setup or what you might have missed during setup. As I said before, chatting on discord server would make things easier for us to troubleshoot.