ihrapsa/KlipperWrt #25: Factory reset

dudzio12, 2021-09-01:
How could I factory reset the device?

I've successfully installed KlipperWrt, managed to connect to it's Wifi hotspot. 
I've connect the device to my local WiFi and restarted.

Now I can see it's connected, but cannot get on it over ssh and local IP with port 81 does not show configuration page.

I've tried to one more time add bz file to reinstall it, but it does not work.

How can I reset the configuration or enable the hotspot one more time or connect to it via ssh?

ihrapsa, 2021-09-01:
> How could I factory reset the device?
>
> I've successfully installed KlipperWrt, managed to connect to it's Wifi hotspot.
> I've connect the device to my local WiFi and restarted.
>
> Now I can see it's [...]

Hi, Hold the reset button for ~6 seconds and it will reset to fresh OpenWrt. Make sure the static ip of the box (192.168.1.1)does not conflict with your router's subnet. If so try changing the the box subnet from 1 to something different

dudzio12, 2021-09-01:
@ihrapsa - it's working, thanks for the tip :)
I've tried pushing down reset button but for more than 10s, which I think just lead it to hard reboot 👍

closed 2021-09-01 by dudzio12