ihrapsa/KlipperWrt #30: Ender3 Upgrade to Klipper

spcasey6907, 2022-05-26:
I have a Creality WIFi Box and have configured with Klipper/Fluidd and appears to be operational, with add digital camera which also works.
Where I am a little scared is the next step creating a 'Printer.cfg' file for the upgrade of the printer's board to klipper. I have read many pages which scare me because you lose control from the attached display to the printer.
Is this true??
How do you tell if my version of the Ender3 will loose control of onboard display?
I am not ready nor was I aware that this was an issue until I got down the rabbit hole.
As I IT Professional I have learned to listen to my instincts....

ihrapsa, 2022-05-26:
> 
If your display is a serial one (like thoe wone from ender 3 V2) it will not work. That display has it's own mcu and can't be controlled by klipper yet. If you've got the classic display it will work with klipper.

spcasey6907, 2022-05-26:
Mine connects via ribbon cable to printer board?!?
Is this what you are saying??

ihrapsa, 2022-05-26:
All displays connect via ribbon cable... Do you even know what printer and mainboard you have?

XoBoR, 2022-05-26:
What he is saying is that TFT displays that have a build in mcu aren't working with klipper, what works is the reprap 12864 style displays, which a lot of creality printers have.

spcasey6907, 2022-05-26:
[Image.jpeg][Image.jpeg]

ihrapsa, 2022-05-26:
> 

You haven't uploaded anything

ihrapsa, 2022-05-26:
Still nothing

closed 2022-10-18 by ihrapsa