ihrapsa/KlipperWrt #31: Latest openwrt factory bin does not seem to be flashing correctly on factory reset creality box from sd card

cobyn, 2022-07-18:
[openwrt_snapshot_wb-01_k5.10.127-factory.bin](https://github.com/ihrapsa/KlipperWrt/blob/main/Firmware/OpenWrt_snapshot/openwrt_snapshot_wb-01_k5.10.127-factory.bin)

ihrapsa, 2022-07-18:
I'm not sure I understand what you mean by factory reset. You were on OpenWrt before? did you get back to srock and now trying to flash again OpenWrt? If you're currently on OpenWrt you need to sysupgrade and use the `sysupgrade bin` not `factory.bin`. If you can't access the box at all, follow the recovery process with the usb stick and `sysupgrade.bin` renamed to `root_uImage` (check README for guide)

cobyn, 2022-07-18:
closing because I was able to solve it

i went from factory -> openwrt -> factory ->  latest openwrt( had issues with original steps to install)

when I put the sd card in with the openwrt_snapshot_wb-01_k5.10.127-factory.bin (it was renamed appropriately) into the box while it was turned on, the machine never restarted and klipperwrt never showed up broadcasting

I decided to try turning the box off and on while the sd card was inserted and that seemed to have worked, I can no see klipperwrt being broadcast.

closed 2022-07-18 by cobyn