ihrapsa/KlipperWrt #33: Creality Camera support?

ffnull, 2022-09-16:
Hello. Im got Sermoon V1 Pro. In it installed creality camera h264. Will it work?

ihrapsa, 2022-09-16:
> Hello. Im got Sermoon V1 Pro. In it installed creality camera h264. Will it work?

Hi, the cam will work. However, that camera is h264 only so the stream will be rtsp not mjpg. RTSP format is extremely eficient in terms of live streaming but it is not web browser compatible. It needs to be converted to mjpg (which requires reencoding and processing power) or other web compatible formats (HLS, WebRTC).
I'm actually looking for a way to convert the stream at the moment.

Also, fluidd/mainsail dev guys are  working at the moment to embedd those streams in the UI as well.

To view the rtsp stream you can use vlc or any rtsp player app

If you're fine with using another app to view the stream I can show you how to set up the rtsp server on the creality wifi box

ffnull, 2022-09-16:
Thnx for reply. Yes its will be great. Im already made hardware mods to printer board to make hotend fan controlable and other fans too.(just cut one pad and solder one wire).

ffnull, 2022-09-16:
by the way. What about? https://github.com/Desuuuu/DGUS-reloaded-Klipper

ihrapsa, 2022-09-16:
> by the way. What about? https://github.com/Desuuuu/DGUS-reloaded-Klipper

What about it?

ffnull, 2022-09-16:
> > by the way. What about? https://github.com/Desuuuu/DGUS-reloaded-Klipper
> 
> What about it?

This branch will be able to us enable touchscreen. Im already use it on another printers with same display.

ihrapsa, 2022-09-16:
Nit sure if that's a statement or it was supposed to be a question. Yes, that's a custom firmware for DWIN T5UID1 screens.

ffnull, 2022-09-16:
> Nit sure if that's a statement or it was supposed to be a question. Yes, that's a custom firmware for DWIN T5UID1 screens.

There both. Custom version of klipper and custom firmware for DGUS screens. Custom klipper updates regulary with official. Can we add support for this branch?

ihrapsa, 2022-09-16:
> > Nit sure if that's a statement or it was supposed to be a question. Yes, that's a custom firmware for DWIN T5UID1 screens.
> 
> There both. Custom version of klipper and custom firmware for DGUS screens. Custom klipper updates regulary with official. Can we add support for this branch?

That's just the screen firmware. I don't understand what you mean/would like. I'm not the developper of the dgus screen firmware or of the klipper fork. You're not making your point clear. 

This is an issue about camera support anyway, so unless you've got extra question about that, I'll close the issue.

ffnull, 2022-09-16:
> That's just the screen firmware. I don't understand what you mean/would like. I'm not the developper of the dgus screen firmware or of the klipper fork. You're not making your point clear.
>
> This is [...]

We did not understand each other. Regarding the camera - I will be glad to instructions.

ihrapsa, 2022-09-16:
> We did not understand each other. Regarding the camera - I will be glad to instructions.

Install `v4l2rtspserver` with `opkg install v4l2rtspserver`, plug the camera and you should be able to watch it on `rtsp://openwrt:openwrt@192.168.x.y:554/stream` Replace the ip with the one of the wifi box.

If you don't want the user and pass, edit `/etc/init.d/v4l2rtspserver` file and comment/delete those lines

ffnull, 2022-09-16:
Thnx!!!!

closed 2022-10-18 by ihrapsa