ihrapsa/KlipperWrt #34: 256MB RAM support

ffnull, 2022-09-18:
Hello again! All works mostly ideal. But i whant to resolder RAM and make it 256mb. Do i need recompile openwrt?

ihrapsa, 2022-09-18:
> Hello again! All works mostly ideal. But i whant to resolder RAM and make it 256mb. Do i need recompile openwrt?

Hi, for ram upgrade you wouldn't have to change the firmware. Make sure it's a compatible chip. Let me know how it goes. Really curious!
Jonah

ffnull, 2022-09-18:
> > Hello again! All works mostly ideal. But i whant to resolder RAM and make it 256mb. Do i need recompile openwrt?
> 
> Hi, for ram upgrade you wouldn't have to change the firmware. Make sure it's a compatible chip. Let me know how it goes. Really curious! Jonah

Its must works without issue. But. Creality Box uses realy old CPU thats supports only DDR2 RAM. So its pretty hard to find 2Gbit ram chip(DDR2). I will inform u

ihrapsa, 2022-09-18:
Yeah, I reckon. Let me know how it goes! 🙂