ihrapsa/KlipperWrt #36: usbip Server support

philipp-andreas, 2022-10-15:
Hello,
usbip server support would be great.

lg Andreas

ihrapsa, 2022-10-15:
Try installing the server package:
`opkg install usbip-server`
Are you thinking of using klipper on a server/separate computer and control the printer over the network? I'm really curious how this goes.

ihrapsa, 2022-10-15:
[UPDATE]
 Apparently the `usbip` openwrt packages require `kmod-usbip` which is a kernel modul and must be precompiled on snapshot OpenWrt images. I can build an image for you if you want.

PeterRob, 2022-10-15:
Alternatively you could try [Virtualhere](https://virtualhere.com) which is a commercial development of usbip, there's a free server available.

philipp-andreas, 2022-10-17:
> [UPDATE] Apparently the `usbip` openwrt packages require `kmod-usbip`, `kmod-usbip-server` and `kmod-usbip-client` which is a kernel modul and must be precompiled on snapshot OpenWrt images. I can build an image for you if you want.

Hello PeterRob
the creality box is to slow for octopi. so i would like to test with octopi on my homeserver and  connect my ender 3 with usbip. an image would be very useful

lg Andreas

Nbrooks639, 2022-10-17:
> Hello PeterRob the creality box is to slow for octopi. so i would like to test with octopi on my homeserver and connect my ender 3 with usbip. an image would be very useful
>
> lg Andreas

could you make this for me please

ihrapsa, 2022-10-17:
[openwrt-usbip-kmods.zip](https://github.com/ihrapsa/KlipperWrt/files/9799782/openwrt-usbip-kmods.zip)
Here are the compiled binaries for the box with the usbip kmods preinstalled (`kmod-usbip`, `kmod-usbip-server` and `kmod-usbip-client`). You also need to install the usbip packages manually:
```
opkg install usbip usbip-server usbip client
```
[Here](https://openwrt.org/docs/guide-user/services/usb.iptunnel) is the OpenWrt guide for extra info on using usb over ip

closed 2022-10-18 by ihrapsa