ihrapsa/KlipperWrt #37: Klipper screen Android and ADXL

BEprints, 2022-11-09:
Not so much a problem but more of a question.

Is it possible to run Klipper screen for android on the Creality box either though wifi or USB? and can the ADXL be used in the second USB port od do you have to do it via the main board?

ihrapsa, 2022-11-09:
Hi, in theory that's possible since the box is just like any other linux device. However, depending on the amount of resource usage klipperscreen server uses ir mifht render the box too unstable. (Its afterall just a device with 580mhz single core and 120mb of ram+swap)

2. Adxl can be uses through usb (indirecrly with an arduino or rpi pico running klipper firmware). However, doijg the resonance tests on the box is not possible/worth it (there's no numpy package available for it and even if you calibrate it with another computer to use the calibrated values on the box it will struggle to keep up with the live resonance calculations when printing

BEprints, 2022-11-09:
I was hoping to be able to use the phones camera as well as having Klipper screen

ihrapsa, 2022-11-09:
You can use the phone's camera with the `IP Webcam` android app then use the mjpg url created in fluidd/mainsail.

There are ways to setup klipper/moonraker etc directly on the phone (replace the box altogether) but you need a phone that's cable of both charging and data transfer through the usb port but I've never tested this.

If you just want to use the phone as a printer screen I would rather use the mobile web ui of fluidd/mainsail or the `mobileraker` app as that's certain to work. 
I haven't tested setting up klipperscreen server on Openwrt so there might be some issues if you try to install using the official script. If I get some free time I might try testing it.

BEprints, 2022-11-09:
Thanks for that I'll look onto it. Just using the phone to run Klipper is looking like a pretty good option.

BEprints, 2022-11-13:
Is there a way to install kipper screen I tried the KIAUH script to install it but when I do sudo apt-get install git -y it comes up with this -ash: sudo: not found and the manual way I do

cd ~/
git clone https://github.com/jordanruthe/KlipperScreen.git
cd ~/KlipperScreen
./scripts/KlipperScreen-install.sh

When I get the that last line it says
-ash: ./scripts/KlipperScreen-install.sh: not found

I know this hasn't been test but I don't mind giving it a go I've managed to get Xserver working on my tablet I just need Klipper screen installed on the box to see how stable it will be.

bitslizer, 2022-11-13:
The script location is minor, the issue is this is a bash script but wrt is using ash

BEprints, 2022-11-13:
Oh I see so I wont be able to try it out. I am currently looking into mobileraker.