ihrapsa/KlipperWrt #38: http://openwrt.local not working

BEprints, 2022-11-09:
I have gone though the entire install process and now can not get access I went with the mainsail option and I have also changed the host name to KlipperWrtMax as I am using two 1 for a ender3v2 and 1 for a max.

I've tried http://openwrt.local http://openwrt.local:81 http://box-ip http://box-ip:81 http://KlipperWrtMax.local http://KlipperWrtMax.local:81 http://klipper.local http://klipper.local:81 http://mainsail.local http://mainsail.local:81

Nothing seems to work apart from LuCI though my networks assigned IP address for the box.

ihrapsa, 2022-11-09:
Hi, if you can't access the box through its ip assigned by the router then something has failed during installation (most likely moonraker). What error do you see when accessing `http://box-ip`?

BEprints, 2022-11-09:
I cant even access http://box-ip it comes up with This site can’t be reached. would it be worth formatting the SD card and trying install again?

ihrapsa, 2022-11-09:
Unplug the sd card with the box off, power on the box then plug the sd card back in after the box finishes booting. After that you can try starting over with the two scripts. First one will take care of the formatting and extroot then the second will install the dependencies and klipper/moonraker

BEprints, 2022-11-09:
I did that and went though the process again and did everything the same as last time. I managed to get it to open mainsail with http://klipperwrtmax.local and it was working fine I then went to find a config file for a Ender 3 Max with CR Touch and now I cant get back into it using http:/klipperwrtmax.local

BEprints, 2022-11-09:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ihrapsa, 2022-11-09:
By `http://box-ip` i mean the actual ip of the box your router assigned, not the words. Replace `box-ip` with the actual ip

BEprints, 2022-11-09:
Oh okay I got you now that's that last photo I can access LuCI with 192.168.1.11:81 but not mainsail

ihrapsa, 2022-11-09:
Did you try the IP without the `81` port? Also, did you change the static lan IP tona different subnet?

BEprints, 2022-11-09:
Just using the IP got me in, I don't think I set a static IP I just followed the instructions and the only thing I did different was change the host name and set a password, I tried to change the SSID name but it wont let me it always times out when I apply and save. 

[image]

ihrapsa, 2022-11-09:
The lan static IP in the box network settings, not in router. Could you check it for me?

BEprints, 2022-11-09:
I'll check when I stop having problems I can't access though home network now only the boxes wifi

BEprints, 2022-11-09:
I really don't know what's going on it connects though http://klippermax.local http://klippermax.local:81 when connected to the boxes wifi then when I try it on my home network it doesn't but sometimes it will connect though the IP the router has assigned it 192.168.1.11 192.168.1.11:81 
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ihrapsa, 2022-11-09:
> No static IP set [image]

That's not what I asked for. Go to network - Interfaces - Edit for the lan interface and show the settings there

BEprints, 2022-11-09:
[image]

BEprints, 2022-11-09:
Thank you for your time and help, It is now 0100 hours where I am so I will have a go tomorrrow and start from scratch by going back to the Creality firmware.

ihrapsa, 2022-11-09:
> [image]

Ok you did change the lan ip subnet to 192.168.2.1 which is fine since ypir router is on 192.168.1.x. However, I'm not sure why you are able to access ot on 192.168.3.1 when connected directly to its hotspot. What did you do there?

> Thank you for your time and help, It is now 0100 hours where I am so I will have a go tomorrrow and start from scratch by going back to the Creality firmware.

You don't need to flash thebstock firmware in order to start from scratch. Just reset the box (hold reset buttom for ~5-6 sec) and  it will revert to fresh openwrt

BEprints, 2022-11-10:
Hi I originally made it 192.168.2.1 then the second time around I just followed the linked video and just did what he did as to take any potential error on my side that from me choosing my own sub net and changing the host name. That screen shot I had just changed it back to 2.1 to see if it would make any difference.

Okay thanks I'll have a go from fresh openwrt I didn't do that last time I just turned the box off took the sd card out turned on the box and once all light were stable put the sd card back in and ran the 2 scripts.

BEprints, 2022-11-10:
It might be the wifi in the box playing up as it wouldn't find any networks when scanning and also just randomly disconnected me.
[image]
[image]
[image]

BEprints, 2022-11-10:
I've managed to get into mainsail though home network at first it came up say This site can't be reached but then reloaded itself into mainsail
[image]

BEprints, 2022-11-10:
So now it's doing the same thing. I put a printer.cfg file one then came back to make some changes and it said connection to moonraker was lost and to retry, nothing was happing so I unplugged the box and powered it back on this time my router gave it a different IP address to what it was before, yesterday it stayed on 192.168.1.11 and today it was 192.168.1.15 and now is 192.168.1.13 and will only connect though that and not http://klippermax.local

[image]
[image]

BEprints, 2022-11-11:
Thanks I will give it a go I got off work a bit earlier today so will try some stuff. It has managed to stay with the .13 IP address today but still can't use a .local address to access mainsail

BEprints, 2022-11-11:
I have reserved 192.168.1.13 for the MAC address of the box in my router

BEprints, 2022-11-11:
The static IP didn't change anything would only connect though the IP and not the .local I have also tried a reset by holding and without the sd card in I didn't put the sd card back in and the wifi name comes up to what I originally changed it to KipperWrtMax with no password and the IP is back at 3.1 But to get into LuCI I need my password and my home network was already set so I might try go back to original firmware and start again

ihrapsa, 2022-11-11:
You don't need to go back to stock to start fresh.  If you hold the reset button and your chanfes are srill there then the reset procedure failed. With or without the sd caed plugged hold the reset button until you see the leds flashing. Wait a minute to reboot and everything should return to default (OpenWrt wifi ap, 192.168.1.1 lan ip etc)

If you manage to install everything and reserve the ip we'll look into the avahi daemon that takes care of the hostname.local access

BEprints, 2022-11-11:
I've held down the rest button for ages the 4th light comes on then second light come on and then they all go out the the process starts over and same result every time. I tried the restore default option that that worked but then when I sent my home network and applied and saved it didn't save and now the boxes wifi doesn't show up so I am going to try get in with a cable.

ihrapsa, 2022-11-11:
Your box is not reseting. You don't have to hold it for ages, just until it reboots

> I tried the restore default option that that worked

What do you mean by that?

BEprints, 2022-11-11:
Ethernet cable worked so I am guessing my rest button isn't working properly

BEprints, 2022-11-11:
I restored back to defaults and have set everything and now it won't show in my home network I will get some screen shots of all my settings I will just need to block out some MAC addresses I really don't know what I am doing wrong.

ihrapsa, 2022-11-11:
> I restored back to defaults and have set everything

What do you mean by that? did you do it manually or?

BEprints, 2022-11-11:
> 

I looked into LuCI and used the option under back up/firmware to restore default as the rest button wasn't doing anything just a power cycle well that's all I think it was doing.

BEprints, 2022-11-11:
I might just have a rubbish box it is Chinese after all, I have a second one I can try I am just away for work at the moment and the second one is at home

ihrapsa, 2022-11-11:
If you want tor reset it without the button, do a sysupgrade from luci. Upload then upgrade with the `sysupgrade.bin`
[image]
Make sure to uncheck the. `Keep settings` option

[image]

BEprints, 2022-11-11:
Okay I'll try that

BEprints, 2022-11-11:
That seems to be working for now, I did the sysupdgrade then unplugged the box removed the sd card and reformatted it I then proceeded with maxing all the changes on the box, host name, passwords, ssid name, home network and terminal I then installed the sd card and ran the scrips all seems to be good at the moment so I will try and access from my phone and save the page to screen and see if that will work reliably.

ihrapsa, 2022-11-11:
If you plan to access the ui from phone with the `hostname.local` it will most likely not work. It's an android limitation afaik as it defaults to google's DNS. This might have changed recently according to [this](https://source.android.com/docs/core/architecture/modular-system/dns-resolver#mdns-local-resolution) but haven't tested it yet. A more stable setup would be to reserve the ip in router then create the chrome web app shortcut on homescreen. On desktop that should not be an issue

BEprints, 2022-11-11:
Thanks, on the phone it is only accessible though box.ip but is working fine on my pc and I was able to create a app on pc and it's been consistently connecting though hostname.local

I find it interesting how the box changed to a different host name without the sd card in so maybe there was some conflicting information as I changed the host name a few times trying to get it to work.

ihrapsa, 2022-11-11:
> I find it interesting how the box changed to a different host name without the sd card in

If you changed the hostname after you setup extroot then those setttings remain on the sd card. If you remove sd card only the settings that were changed on the internal NOR chip apply.

I'm glad you fixed your issues

closed 2022-11-11 by ihrapsa