ihrapsa/KlipperWrt #39: Install issue with 2_script_manual.sh

the-moog, 2022-11-30:
Issue with the pip install ... lmdb ... line in scripts/2_script_manual.sh

Error text
```bash
Collecting lmdb
  Using cached lmdb-1.3.0.tar.gz (881 kB)
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> See above for output.

  note: This error originates from a subprocess, and is likely not a problem with pip.
  Preparing metadata (setup.py) ... error
error: metadata-generation-failed
```
The above error did not highlight the actual problem.  Running pip with -v did.

It seems lmdb is patched and built in situ.  It needs `pip install wheel` and `opkg install patch` to install.

ihrapsa, 2022-12-02:
Hi, did you manage to bypass that with `patch` and `wheel`?

the-moog, 2022-12-03:
@ihrapsa 
The install of lmdb fails because it's pip install invocation compiles some Python C extensions on the fly then installs them (using wheel).  For some reason in doing that it applies a patch to the source, so I guess the author of that component never finished the job on the Python version in that OpenWRT release.  The binutils patch tool is simply missing on the Creality Box port of OpenWRT so it tries to run it, silently fails then does not compile. Sadly pip's error reporting buried this and reported the egg-info error instead.  Also the CrealityBox OpenWRT is not in the main OpenWRT repo so it's lagging somewhat meaning lots of updates fail due to kernel compatibility.  Maybe I will try and find time to fix that.

long story short (it took hours for me to work this one out) I did something like
sudo opkg install patch
sudo pip install wheel

(IIRC, it was a few days back and my memory is not great - I may have updated pip too)

ihrapsa, 2022-12-03:
Glad you fixed it eventually. Yeah, unfortunately the box is not officially supported yet. Some packages can't be built on the target (mostly Python C extensions) and because of that I always prefer to use openwrt repo prebuilt python modules. If that's not an option I crosscompile. That lmdb pacakge used to be installed from a crosscompiled `.ipk` I put [here ](https://github.com/ihrapsa/KlipperWrt/blob/main/packages/python3-lmdb%2Bstreaming-form-data_packages_1.0-1_mipsel_24kc.ipk) but switched back to pip when the target compilation issues resolved. The package is pretty old and comes bundled with some other python package that had similar issues.