ihrapsa/KlipperWrt #40: klipperwrt.local not known

josephchanhk, 2022-12-02:
Hi,

I am KlipperWrt v2.2  Finished following two script.
 
Download and execute the 1_format_extroot.sh script:
Download and execute the 2_script_manual.sh script:

But Klipperwrt.local is not know. cannot access the Klipper client

ihrapsa, 2022-12-02:
Try the ip instead

josephchanhk, 2022-12-02:
May I know the IP should be the 3D print IP?
So if I not yet connect the 3D print. "klipperwrt.local" should be not work?

ihrapsa, 2022-12-02:
The ip of your box that your router assigned. It should be in router dhcp list next to openwrt hostname

josephchanhk, 2022-12-02:
my router IP is 10.234.1.1
In the install process. I login the "WIFI box" ip is 10.234.1.219
so you mean ip should be same 10.234.1.219 ?
How about the printer UI interface? Is it need connect to moonraker?

ihrapsa, 2022-12-02:
You’re confusing some stuff here. Your box static ip when first truing to setup klipper wrt on it is 192.167.1.1. After you join your home network at the wifi setup step your box ip will change to the one your router will tell. Using that ip your box will be accessible on your home network from any device. The printer ui interface uses same ip as any other interface on the box

josephchanhk, 2022-12-02:
Yes when first setup "PC" directly connect wifi is Klipperwrt directly to "crealty wifi box" is 192.168.1.1
After input my home network wifi name and password to "BOX"
I disconnect klipperwrt, and then connect normal to my home netwrok. 
At this time "BOX" also connect to my home network. And the "BOX" ip will not be 192.168.1.1
I use [wireless network watcher] can see all network device ip.

My home network router IP is 10.234.1.1
The "BOX" IP is 10.234.1.219
In the install process. I login the "BOX" by ip 10.234.1.219

I only confuse how I can go to Fluidd/Mainsail UI.

Thanks

ihrapsa, 2022-12-02:
Access http://10.234.1.219 fluidd/mainsail should be at that address if everything installed correctly. At http://10.234.1.219:81 should be luci UI

josephchanhk, 2022-12-02:
Thanks a lot. Yes this ip can luci UI.
I am sorry, I confuse with luci UI and " Fluidd/Mainsail UI. "
luci UI only display the "BOX" setup. 
I do not know where can setup the 3D printer.

josephchanhk, 2022-12-02:
Yes I can access to the terminal window. 
where can I find instal logs?

Thanks

ihrapsa, 2022-12-02:
> Yes I can access to the terminal window. where can I find instal logs?
> 
> Thanks

It's the entire terminal output you need to copy paste in a text document. The script does not create any logs.

josephchanhk, 2022-12-02:
BusyBox v1.35.0 (2022-07-05 19:23:44 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r20027-586c2426d9
 -----------------------------------------------------
root@KlipperWrt:~# cd ~
root@KlipperWrt:~# wget https://github.com/ihrapsa/KlipperWrt/raw/main/scripts/1_format_extroot.sh
--2022-12-02 15:31:03--  https://github.com/ihrapsa/KlipperWrt/raw/main/scripts/1_format_extroot.sh
Resolving github.com... 20.205.243.166
Connecting to github.com|20.205.243.166|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/ihrapsa/KlipperWrt/main/scripts/1_format_extroot.sh [following]
--2022-12-02 15:31:04--  https://raw.githubusercontent.com/ihrapsa/KlipperWrt/main/scripts/1_format_extroot.sh
Resolving raw.githubusercontent.com... 185.199.110.133, 185.199.111.133, 185.199.108.133, ...
Connecting to raw.githubusercontent.com|185.199.110.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2596 (2.5K) [text/plain]
Saving to: '1_format_extroot.sh.2'

1_format_extroot.sh.2             100%[=============================================================>]   2.54K  --.-KB/s    in 0.001s  

2022-12-02 15:31:04 (2.71 MB/s) - '1_format_extroot.sh.2' saved [2596/2596]

root@KlipperWrt:~# chmod +x 1_format_extroot.sh
root@KlipperWrt:~# ./1_format_extroot.sh
 
This script will format your sd card and make it extroot
 
   ###################################################
   ## Make sure you've got a microSD card plugged!  ##
   ###################################################
 
Press [ENTER] to continue...or [ctrl+c] to exit
This script will format your sdcard. Are you sure about this? [y/n]: y
umount: can't unmount /overlay: Resource busy
mke2fs 1.46.5 (30-Dec-2021)
/dev/mmcblk0p1 is mounted; will not make a filesystem here!
yes: (null): Broken pipe
 
cp: '/overlay/.' and '/mnt/.' are the same file=====>    ](93%)
Making extroot...     [=================================>](100%)
Please reboot then run the second script!
root@KlipperWrt:~#

josephchanhk, 2022-12-02:
BusyBox v1.35.0 (2022-07-05 19:23:44 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r20027-586c2426d9
 -----------------------------------------------------
root@KlipperWrt:~# cd ~
root@KlipperWrt:~# wget https://github.com/ihrapsa/KlipperWrt/raw/main/scripts/2_script_manual.sh
--2022-12-02 15:33:48--  https://github.com/ihrapsa/KlipperWrt/raw/main/scripts/2_script_manual.sh
Resolving github.com... 20.205.243.166
Connecting to github.com|20.205.243.166|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/ihrapsa/KlipperWrt/main/scripts/2_script_manual.sh [following]
--2022-12-02 15:33:48--  https://raw.githubusercontent.com/ihrapsa/KlipperWrt/main/scripts/2_script_manual.sh
Resolving raw.githubusercontent.com... 185.199.110.133, 185.199.109.133, 185.199.108.133, ...
Connecting to raw.githubusercontent.com|185.199.110.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 12632 (12K) [text/plain]
Saving to: '2_script_manual.sh.2'

2_script_manual.sh.2              100%[=============================================================>]  12.34K  --.-KB/s    in 0.004s  

2022-12-02 15:33:49 (3.34 MB/s) - '2_script_manual.sh.2' saved [12632/12632]

root@KlipperWrt:~# chmod +x 2_script_manual.sh
root@KlipperWrt:~# ./2_script_manual.sh
 
   ################################################
   ## Did you execute 1_format_extroot.sh first? ##
   ################################################
 
Press [ENTER] if YES ...or [ctrl+c] to exit
 
This script will download and install all packages form the internet
 
   #####################################
   ## Make sure extroot is enabled... ##
   #####################################
 
Press [ENTER] to check if extroot is enabled ...or [ctrl+c] to exit
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 6.3M      6.3M         0 100% /rom
tmpfs                   256.0M    112.0K    255.9M   0% /tmp
/dev/mmcblk0p1            7.3G      1.0G      5.9G  15% /overlay
overlayfs:/overlay        7.3G      1.0G      5.9G  15% /
tmpfs                   512.0K         0    512.0K   0% /dev
 
   ############################################
   ## Is /dev/mmcblk0p1 mounted on /overlay? ##
   ############################################
 
Press [ENTER] if YES... or [ctrl+c] to exit
 
   ########################################################
   ## Make sure you've got a stable Internet connection! ##
   ########################################################
 
Press [ENTER] to Continue ...or [ctrl+c] to exit
 
#################
###   SWAP    ###
#################
 
Creating swap file
dd: can't open '/overlay/swap.page': Text file busy
Enabling swap file
[... 817 lines cut ...]
Package usbutils (014-1) installed in root is up to date.
 
########################
###  Fixing logs...  ###
########################
 
Creating system.log...
 
Installing logrotate...
 
Package logrotate (3.17.0-2) installed in root is up to date.
 
Creating cron job...
 
 
Creating logrotate configuration files...
 
 
#################
###   Done!   ###
#################
 
Please reboot for changes to take effect...
...then proceed configuring your printer.cfg!
root@KlipperWrt:~#

josephchanhk, 2022-12-02:
Oh sorry, I over look your reply.
I should use http://10.234.1.219/   without :81

This issue solve

closed 2022-12-02 by ihrapsa