ihrapsa/KlipperWrt #41: Webcam is not working with mjpg-streamer

kgsrirama, 2022-12-06:
Hi,

I installed openwrt on the Creality web box and was able to successfully able to run klipper and do prints. I was trying to install mjpg-steamer and webcam, but all methods failed. BTW, I am able to use ffmpeg to record video and download it.

running v4l2-ctl --list-devices
`Integrated Camera: Integrated C (usb-101c0000.ehci-1.2):
	/dev/video0
	/dev/video1
`

ihrapsa, 2022-12-06:
Hi,
`mjpg-streamer` is installed by default. OpenWrt uses `/etc/config/mjpg-streamer` as the configuration file. The server is restarting every time you plug a video device. To troubleshoot do `logread -f` and unplug-> replug the usb webcam. `logread -f terminal should output some logs. The mjpg-streamer server should be on `http://<ip>:8080`

kgsrirama, 2022-12-06:
Sorry forgot to mention earlier, checked those too. Tried toggling option enabled to 1 and device to /dev/video1 /etc/config/mjpg-streame but nothing changed

ihrapsa, 2022-12-06:
What webcam are you using?

kgsrirama, 2022-12-06:
It is Enter Device 004: ID 13d3:784b FLH-BY-0309-200514 Integrated Camera with SPCA2072A chip in it. I use it in my ubuntu laptop. BTW I can see it working in the commandline mjpg-streamer in klipper/ender3 box too

Sorry for split messages (used to Ctrl+enter in another program to move to next line)

This is the only output after accessing http://ender3.lan:8080/webcam/stream.html which is unrealted to above error
Tue Dec  6 21:27:30 2022 daemon.err odhcpd[1663]: Failed to send to ff02::1%lan@br-lan (Address not available)

kgsrirama, 2022-12-06:
Sorry for the trouble, I am able to solve the issue. It was due to a conflict in port for 8080. I guess KlipperWRT pushes openwrt to port 8080 in nginx configuration. But I moved openWRT to 8000 using luci and tried using 8080 for mjpg-streamer. However, 8080 configuration remained in nginx and access url to mjpg-streamer was always redirected by cgi-bin/luci. 

I solved it temporarily moving mjpg-streamer to port 8090

closed 2022-12-06 by kgsrirama

array81, 2023-10-15:
I have a similar problem with my webcam plug on USB2.

This is my code:
```
config mjpg-streamer 'core'
        option enabled '0'
        option input 'uvc'
        option output 'http'
        option device '/dev/video0'
        option resolution '640x480'
        option yuv '0'
        option quality '80'
        option fps '5'
        option led 'auto'
        option www '/www/webcam'
        option port '8080'
        #option listen_ip '192.168.1.1'
        #option username 'openwrt'
        #option password 'openwrt'
```