ihrapsa/KlipperWrt #44: Automatic install is not working.

3dprintpt, 2023-01-11:
After installing with the automatic scripts, I always get an error related to nginx when trying to reload the page.
502 Bad Gateway
nginx/1.21.3

http://192.168.1.208
No moonraker connection. Please check moonraker status and / or refresh.

Maybe this has something to do with newer version of moonraker, not sure.
Can anyone help with this?

Thank you in advance.

SmallTownSound, 2023-01-13:
Noticed going thru auto install that GCC package is not loading, klipper and moonraker will not work without it.

3dprintpt, 2023-01-13:
> Noticed going thru auto install that GCC package is not loading, klipper and moonraker will not work without it.

any idea on how to solve it?

SmallTownSound, 2023-01-14:
No response here or on discord.   Must be a recent problem, a lot of people on Discord mentioning it.

yetiwerx, 2023-01-14:
What happens when you install gcc first in WRT Software then run scripts?

3dprintpt, 2023-01-15:
> What happens when you install gcc first in WRT Software then run scripts?
> -- Randy Kopel Head Tinkerer Yeti Design Werx, LLC 2621 2nd St Entiat, WA 98822 Ph: (509)423-3500

will try tomorrow and post here.
Will see If I can install GCC. Yesterday I couldn't, package unavailable...

SmallTownSound, 2023-01-15:
Just tried something and it worked:
ran automatic install scripts #1,   reboot
run:  opkg update
run:  opkg install gcc
then run install script #2, then reboot.
After that, I copied the printer.cfg I created to use printer on Sonic Pad, using same klipper firmware on printer, only changed [MCU}  to  'serial: /dev/ttyUSB0'
It now works.  It does take a bit for moonraker and klipper to load after reboot, so if you get erros, give it a minute or 2.
now only error i'm getting is  'GCode path received from Klipper does not exist: /gcode_files'
All fixed.   changed to 'path: /root/printer_data/gcodes' in printer.cfg.
Just started a print from moonraker plugin.... 

Don

3dprintpt, 2023-01-16:
> Just tried something and it worked: ran automatic install scripts #1, reboot run: opkg update run: opkg install gcc then run install script #2, then reboot. After that, I copied the printer.cfg I [...]

This worked flawlessly.
Thanks a lot...

5vicius, 2023-03-24:
Sorry in advance for the long post, but I think that getting the logs would be a way to find answers

echo805, 2023-06-29:
Having same issue then prompted to reboot after I reboot terminal says connection closed and no more access through ip or wrt.local

naldarn, 2023-10-26:
Still having the same issue.  Looking at @5vicius error log, looks like /libpython3.10.so.1.0 is causing the problem he was having , but mine all stem from gcc and make not being available to run the make menuconfig command.

closed 2024-05-20 by 3dprintpt