ihrapsa/KlipperWrt #53: connection error

Randomguyon2, 2024-03-30:
when i connect to the wifi box, I cannot connect to my home wifi on my laptop, what do I do? thanks.

smit9418, 2024-04-15:
need more info. are you saying that when the wifi box is connected to your main ssid, your laptop then will not connect to the main ssid? these are completely independent network devices and should not affect each other at all. i believe your saying both are connected but the wifi box isnt visable on the network. try running advanced IP scanner on the laptop when both are connected to wifi.

shivajiva101, 2024-04-16:
I think they were simply wondering how to setup the box to connect to the main router and configure the firewall for access because they didn't read the manual. Can't use the laptop wifi to connect to both at the same time.

smit9418, 2024-04-16:
> I think they were simply wondering how to setup the box to connect to the main router and configure the firewall for access because they didn't read the manual. Can't use the laptop wifi to connect to both at the same time.

the laptop and wifi box should be connected to the main SSID at the same time. that is what i stated. what are you talking about?

shivajiva101, 2024-04-16:
They specify wifi box and home wifi so I'm making the assumption they are using their laptop to connect to either the main router AP or the Creality box AP and obviously not both at the same time. In that scenario they haven't setup wifi access on the Creality box to their main router and adjusted the firewall to pass through the packets so they only have the option of connecting to the Creality box AP or their main routers AP rather than the method described in the readme where the box is made accessible to any device connected to the main router via a dhcp client wireless connection to their main router.  I was pointing out that it may not be as complicated a scenario as the one you'd assumed, understandable given the lack of detail in the post, and they may simply not have read/understood the instructions in the readme given the lack of technical information in their post.

Beyond all of this is the fact this repo is slightly outdated and currently not maintained so there are other issues related to using this repository that are not readily overcome. For this reason I maintain a fork of KlipperWrt that can be found [here](https://github.com/shivajiva101/KlipperWrt) and my releases are hooked into the KlipperWrt discord channel [here](https://discord.gg/ZGrCMVs35H) under github-releases