ihrapsa/KlipperWrt #54: Failed to download packages

wydget, 2024-12-09:
I was attempting to install this and on the second step, I began receiving errors that it could not download the package list. I went to address and received 404 errors. I then traced it back to OpenWrt and the packages are missing. I am attaching a text doc that has my entire output. Can you please assist?
[Failed Install.txt](https://github.com/user-attachments/files/18054716/Failed.Install.txt)

thank you

sgarcin, 2024-12-09:
you can upograde and install by following this fork https://github.com/shivajiva101/KlipperWrt

sgarcin, 2024-12-09:
be sure to follow the indication to be in the good version : see releases 
IMPORTANT: Click this link to the [firmware](https://github.com/shivajiva101/KlipperWrt/tree/v3.4/Firmware/OpenWrt_snapshot) to ensure you are synchronised to the scripts and installation instructions and make sure you stay on the v3.4 tag!

OpenWrt 23.05.4 r23001+1011-38c150612c
LuCI openwrt-23.05 branch git-24.233.62451-74b688c
Kernel version: 5.15.162

sgarcin, 2024-12-09:
you can flash directly from the /cgi-bin/luci/admin/system/flash url, the new sysupgrade.bin from the previous link
no need to flash from micro sd card again